keijiro/Rsvfx #4: Multiple real sense camera

antoniohof, 2019-03-15:
Is it possible to use multiple real sense cameras in this project connected to the same PC?

Thank you

Antonio

bryanrtboy, 2019-03-18:
I know this is not designed for Mac, but I was able to get all of the example scenes working except for Particles. That seems to be the only example using quad output, but I have not been able to figure out why it's not working. There is no emmision of any particles on Play or in the Editor.

keijiro, 2019-03-18:
@bryanrtboy Is it related to the topic? ("Multiple real sense camera")

If it's not relevant, please repost it as a new issue.

keijiro, 2019-03-18:
@antoniohof I only have a single RealSense device so I have never tried it, but I think you can use multiple devices with specifying serial numbers in Rs Device components ("Requested Serial Number" property).

closed 2019-03-18 by antoniohof