keijiro/Rsvfx #5: Particle Demo not working on Mac

bryanrtboy, 2019-03-18:
I know this is not designed for Mac, but I was able to get all of the example scenes working on Mac OSX except for Particles. 

I figured out what is wrong, the Quad Output module requires a Scale.X.random or Scale.X. from Map block. Without that block, the effect works while editing the VFX graph, but does not compile or work in play mode.

keijiro, 2019-03-25:
This doesn't reproduce with the latest version.

I have updated HDRP and VFX Graph, so it might be dependent on the version of these packages.

keijiro, 2019-03-25:
I'm closing the issue now. Please feel free to reopen when it still reproduces on your side.

closed 2019-03-25 by keijiro

bryanrtboy, 2019-03-28:
I'm still getting this error on the Mac. Using a fresh download, 2019.1.0b9  with VFX 5.10.0. The Particle and Rain demo only work while editing the VFX graph. When I use Force Runtime Mode, there are no particles emitting, which is the same as closing the VFX and hitting Play.

The fix is to add a Map block, Set Scale with Map.XYZ with Random selected in the pulldown. It emits using Force Runtime and on Play. It seems like this may be a VFX bug and Mac OS X or something to do with how those two scenes are set up.