keijiro/Rsvfx #8: Is incompatible with Unity 2019.2.0a9?

comoc, 2019-04-02:
I tried to open 688fefc version of Rsvfx with Unity 2019.2.0a9 on Windows 10.
However, I faced some Package related errors.
Is Rsvfx incompatible with Unity 2019.2.0a9 so far?

Best regards,

keijiro, 2019-04-03:
I'm pretty sure that there are some compatibility breaking changes in HDRP and VFX Graph. I'll look into it later.

nickcoppins, 2019-09-19:
I'm wondering if there was any update to this? 
Rsvfx in 2019.1.6f1 was working fine. 

Trying in 2019.2.6f1 and i'm find errors relating to the float3/float4 types not being found in the unity.mathematics namespace?

keijiro, 2019-09-19:
This issue was fixed in the latest commit. https://github.com/keijiro/Rsvfx/commit/c289f3bc302d57b65f09108065de47191be8b1b5

keijiro, 2019-09-19:
I'm closing this issue now. Please feel free to reopen for further questions.