keijiro/Rsvfx #9: Erratic PointCloudMap (on iGPU)

Pithone, 2019-04-03:
Hi, I'm experiencing various issues in both 2018.3.11 and 2019.1.0 versions.
The PointCloudMap is incorrect as visible on the GIF; as the RealSense SDK and VFX both work otherwise, it seems to me related to the type of graphic card (on my system it's a Intel iris 650).

[image]

By any chance the creator or any user of this repository can verify this theory by checking if it works on his recent iGPU by Intel?

PS: the ColorMap doesn't work too, but that's less relevant for me and probably related to the core issue on PointCloudMap.

keijiro, 2019-04-05:
I suspect that the iGPU you're using doesn't support half precision floating point texture formats. I made a small utility to check the graphics capability. Could you give a try to it? Please download the package from [here](https://github.com/keijiro/GfxCapsView/releases/download/v0.1.0/GfxCapsView.unitypackage) and import the package to the project, then navigate Help -> GfxCapsView -> Graphics Formats. It enumerates the available graphics formats, so please copy them and paste in your reply.

Pithone, 2019-04-05:
Hi, thank You, I will test the utility this weekend (don't have the D415 with me now).

May i bring to your attention that the Kinect repository made by Roel based on your work experienced the same issue with iGPU, but yesterday it got fixed by the author by modifying the compute shader of the PointCloud in two ways:

1. "Buffer" doesn't work on intel chip, so he changed it to StructuredBuffer"
2. The floats had been changed to decimal value (1 => 1.0) 

Now that repo works (on intel UHD 630 and 650).
I'm hopeful this might come as a potential solution also for your repository!
Nonetheless expect a reply on the results from utility you made!

keijiro, 2019-04-05:
Thank you for providing the information. I managed to reproduce the issue by enabling iGPU of my PC. I also confirmed that the issue can be resolved by changing `Buffer` to `StructuredBuffer`. Actually I met the same issue in some other projects but forgot to apply the fix to this project.

I'm closing this issue now. Please feel free to reopen if you find further issues.

closed 2019-04-05 by keijiro