keijiro/Rsvfx #11: doesn't work with D435i

phields, 2019-04-12:
[image]
Depth camera is on, but it's output nothing.
(Unity Version2019.1.0b10,everything is default)

keijiro, 2019-04-12:
Thanks for reporting the issue.

- Is there any message on the console view?
- Could you update Unity to 2019.1.0f1? I'm sure that the RealSense SDK doesn't care about such a small version difference but HDRP and VFX Graph do.

phields, 2019-04-12:
@keijiro 
Thanks for reply.
There is no any console view(0 message), it looks like 2019.1.0f1 failed just like 2019.1.0b10.

phields, 2019-04-12:
I think the reason is there is no render texture output（color and position）.
But the RealSenseSDK2.0's demo worked normal..

keijiro, 2019-04-13:
How about decreasing the framerate in the Rs Device profile to 30? D415 supports 60 fps but I don't know if D435 supports it.

phields, 2019-04-13:
@keijiro I tried, 30 or 60fps both working in RealSenseSDK demo, but still no rendering in the Rsvfx.
Dkvfx is work for me, so that might be not the HDRP and VFX Graph's problem.

phields, 2019-04-13:
https://github.com/IntelRealSense/librealsense/issues/2455
https://github.com/IntelRealSense/librealsense/issues/3474
I found something useful. 
It looks like the color and depth frame is not align on D435i ?(with same width&height)

bdaver, 2019-04-13:
> IntelRealSense/librealsense#2455
> IntelRealSense/librealsense#3474

Try changing regional settings on your pc. Especially date and number format to united states.  

I spent days to understand why vfx rendering stops. It cripples the shader ehen you compile. It is a rarely known bug on unity. Timeline animations suffer from the same problem as well. When you try to rotate game objects, it goes to billions in degrees.

phields, 2019-04-13:
> Try changing regional settings on your pc. Especially date and number format to united states. I spent days to understand why vfx rendering stops. It cripples the shader ehen you compile. It is a [...]

I tried.
Still not working:(
vfx rendering nothing because PointCloudBaker doesn't output textures

phields, 2019-04-13:
[image]

keijiro, 2019-04-13:
How about upgrading the firmware? I hardly think it helps but should be tried once.

phields, 2019-04-13:
> 
> 
> How about upgrading the firmware? I hardly think it helps but should be tried once.

 Tried, I decided to return the D435i to Amazon.

kromond, 2019-05-28:
Hi,  I have the D435 (not with the i) and also is not working.  The console says this
```
ExternalException: rs2_pipeline_start_with_config(pipe:000001A2A40CE5B0, config:000001A2A40CFC90)
Rethrow as Exception: Couldn't resolve requests
Intel.RealSense.ErrorMarshaler.MarshalNativeToManaged (System.IntPtr pNativeData) (at <39ff202c67d64a63acbba4ae542e7167>:0)
Intel.RealSense.Pipeline.Start (Intel.RealSense.Config cfg) (at <39ff202c67d64a63acbba4ae542e7167>:0)
RsDevice.OnEnable () (at Assets/RealSenseSDK2.0/Scripts/RsDevice.cs:71)
```

For the T265 I see similar
```
ExternalException: rs2_pipeline_start_with_config(pipe:000001A2A28AA030, config:000001A2A28AA1D0)
Rethrow as Exception: No device connected
```
D415 works as expected.  For all three I'm getting the Serial number from the realsense viewer and putting it into Requested Serial Number.  Are there any other steps required?

keijiro, 2019-05-29:
@kromond Regarding T265, it has to be activated before the other RealSense devices.

https://github.com/IntelRealSense/librealsense/issues/3734

To do this with Capture.unity example, please try the following steps:

1. Stop playing.
2. Deactivate RealSenseVGA60 game object.
3. Start playing.
4. Activate RealSenseVGA60 game object.

I'm trying to find a better solution to this issue.

kromond, 2019-05-29:
thank you @keijiro, T265 example working now.  Also D435 works fine when I create a new RsDevice

keijiro, 2019-06-18:
Finally I bought a D435i device and confirmed that it works without any problem on my side.

I'm closing this issue now. Please feel free to reopen if it still reproduces.

closed 2019-06-18 by keijiro

mhkim0720, 2022-06-15:
It's simple. Change the color and depth frame rate to 30 and the 435i will work as well.