keijiro/Rsvfx #12: Enabling RealSense on OSX

DarkMio, 2019-04-16:
Your examples work well on OSX, once the right dependencies have been bound.

For that, I merely pulled the most recent ``librealsense2`` via brew and copied the files:

```
$ brew install librealsense2
$ cp /usr/local/Cellar/librealsense/2.20.0/lib/librealsense2.dylib $PROJECT_PATH/Assets/RealSenseSDK2.0/Plugins/librealsense2.bundle
```

I suspect you've used an older RealSenseSDK2.0 unity package, which seems to be incompatible. Due to this, I removed all other files in Plugins, downloaded the ``2.20.0`` .unitypackage from here: https://github.com/IntelRealSense/librealsense/releases/tag/v2.20.0 and imported only the ``Plugins`` part. After restarting the editor all your samples work fine with a D435.

D435i seems not to work currently due to incompatibility issues with ``librealsense2`` at time of writing.

Thanks!

keijiro, 2019-04-16:
I'm using v2.19.0

https://github.com/keijiro/Rsvfx/commit/5efab8b65bb6bb76bd510469806bb61c69facc1b

So, first of all, could you try installing v2.19.0 via brew? I'm not sure if this exact version is available on brew though.

DarkMio, 2019-04-16:
That one doesn't exist on homebrew apperantly. Building it is rather trivial (https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md) - so I did that.

I had some issues with `libusb`, which was fixed by doing this when compiling:

```
$ make LIBRARY_PATH=/usr/local/lib
$ cp librealsense/librealsense_build/build/librealsense2.2.19.0.dylib $PROJECT_PATH/Assets/RealSenseSDK2.0/Plugins/librealsense2.bundle
```

Resetting the project and reimporting prior yields the desired result, too: [image]  
(My work telephone)

so it looks like ``librealsense2.08`` and onwards should have enough capabilities to work with your repository. If you want my compiled bundle as PR, I could do that, too.

keijiro, 2019-04-17:
Thanks for the clarification. Now it's clear that the issue was caused by a version mismatch. I'll try updating the plugin to 2.20.

keijiro, 2019-04-17:
I updated the SDK in the latest commit and verified that it works on my MacBook.

DarkMio, 2019-04-17:
Thanks, very nice!

Have a nice day.

keijiro, 2019-04-19:
Now I'm closing this issue. Please feel free to reopen for further problems.

closed 2019-04-19 by keijiro