keijiro/Rsvfx #15: RSDevice Pipeline.Start Error

petecleary, 2019-04-28:
I have an issue where the RSDevice throws an error calling Pipeline.Start 

ExternalException: rs2_pipeline_start_with_config(pipe:000001B822500F50, config:000001B822501070)
Rethrow as Exception: Couldn't resolve requests
Intel.RealSense.ErrorMarshaler.MarshalNativeToManaged (System.IntPtr pNativeData) (at <1ab02197d096429db5e25d131c03a77b>:0)
Intel.RealSense.Pipeline.Start (Intel.RealSense.Config cfg) (at <1ab02197d096429db5e25d131c03a77b>:0)
RsDevice.OnEnable () (at Assets/RealSenseSDK2.0/Scripts/RsDevice.cs:71)

I am running a D435 and the camera works every time within the same project with the RealSense Scenes but has only worked once with your RealSense Prefab.  Once working I could run any of your scenes without issue.
I am running Unity 2019.1.0f2 and have the latest RealSense SDK 2.21.0 running with the latest 5.11.6.200 firmware installed onto the camera.

petecleary, 2019-04-28:
I have found the issue, it was not connected to the RSDevice in that the script to move the camera in your scene's was missing.  This caused the error to show and the system to stop.
I have it working with the setup above in your project and another one.
Thank you for such awesome work.

keijiro, 2019-04-29:
I'm closing this issue now. Please feel free to reopen for further questions.

closed 2019-04-29 by keijiro

sophiedixon, 2019-12-21:
Hi @petecleary 
I'm getting the same error you had, but don't understand what you mean by
> the script to move the camera in your scene's was missing
Could you elaborate? Many thanks!