keijiro/Rsvfx #16: Releasing  vertex data

FernandoMonroy, 2019-05-03:
Thanks for such a great tool.

I review the SDK RS code and you make your own improvements. I assume that the MemCopy native function is now embedded on the native side. 

How you release Vertex data on your RS SDK version, previosly you need to call .Release().

keijiro, 2019-05-04:
There is no modification in the RealSense SDK. I just implemneted a custom component that receives data from a RealSense device using the SDK components. Why did you think that I made improvements on the SDK?

keijiro, 2019-05-15:
I'm closing this issue now. Please feel free to reopen for further questions.

closed 2019-05-15 by keijiro

FernandoMonroy, 2019-05-15:
Sorry for the lack of communication.

I test your implementation with the Points component and I get around 3 to 5 frames up in contrast with the original Intel Real Sense SDK.