keijiro/Rsvfx #17: Trying with Orbbec Astra Depth Camera

xubi1993, 2019-05-21:
Hello,
Hope you are doing fine. I am trying to make this code work with the orbbec astra depth camera. But the unity crashes when it try to run the Unsafe Utility code when setting the buffer data. I have tried to set the data by using unity's own set data by converting the intptr to an array but the output texture comes out empty with that. Any help would be much appreciated.
Thanks! 

/////Code//////
        void UpdateColorData(Astra.ColorFrame frame)
        {
            if (frame.DataPtr == System.IntPtr.Zero) return;

            var size = frame.Width * frame.Height;

            if (_colorBuffer != null && _colorBuffer.count != size)
            {
                _colorBuffer.Dispose();
                _colorBuffer = null;
            }

            if (_colorBuffer == null)
                _colorBuffer = new ComputeBuffer(size, 4);

            //Me trying to set color buffer data using set data...
            byte[] managedArray = new byte[size];
            Marshal.Copy(frame.DataPtr, managedArray, 0, size);
            _colorBuffer.SetData(managedArray);
            _dimensions = new Vector2Int(frame.Width, frame.Height);

            //Usafe utility code that causes the unity crash...
            //UnsafeUtility.SetUnmanagedData(_colorBuffer, frame.DataPtr, size, 4);
        }

keijiro, 2019-05-22:
The forth argument of `SetUnmanagedData` is a stride of an array. In your case, `size` has a byte size of the array, so I think it should be `1`. Or the third argument should be `size / 4`.

By the way, have you read the comment of `SetUnmanagedData`?

https://github.com/keijiro/Rsvfx/blob/master/Assets/Rsvfx/Runtime/Utility.cs#L17

I think you shouldn't use it. Or do it at your own risk.

keijiro, 2019-05-22:
I'm closing this issue now. Please feel free to reopen for further problems.

closed 2019-05-22 by keijiro

xubi1993, 2019-05-22:
Thank you for the reply and yes I have read the comment in the SetUnmanagedData that is why I am trying to use the Marshel copy or some other method to set the values of the compute shader but so far nothing seems to be working. Unity itself provides "_colorBuffer.SetData(System.Array);" this function to set the shader data but it takes an array and not the intPtr. Any idea how to go about it?

keijiro, 2019-05-24:
I think you should match the data element size; If you're sending `byte[]`, the compute shader should receive it as `StructuredBuffer<byte>` or something like that. However, as far as I know, HLSL doesn't support `byte`, so it might not work.

If the data is stored as a byte array, how about using `Texture2D` and `LoadRawTextureData`? It might be easier than using a compute buffer.