keijiro/Rsvfx #22: Is incompatible with Unity 2019.1.7f1

fdiba, 2019-06-19:
Hello, I get the error below while using  Unity 2019.1.7f1 :

ExternalException: rs2_pipeline_start_with_config(pipe:000002821EF7BD60, config:000002821EF7C1E0)
Rethrow as Exception: Couldn't resolve requests
Intel.RealSense.ErrorMarshaler.MarshalNativeToManaged (System.IntPtr pNativeData) (at <39ff202c67d64a63acbba4ae542e7167>:0)
Intel.RealSense.Pipeline.Start (Intel.RealSense.Config cfg) (at <39ff202c67d64a63acbba4ae542e7167>:0)
Rsvfx.CombinedDriver.Start () (at Assets/Rsvfx/Runtime/CombinedDriver.cs:114)

What can I do ?

fdiba, 2019-06-19:
Ok problem solved. I don't know why but it's only working with one on my USB port even if the real sense viewer works with all of them.

closed 2019-06-20 by keijiro

Sambhav-Viztech, 2020-11-26:
Thank you for your swift response but still no luck on my end. Here is my error log:-
ExternalException: rs2_pipeline_start_with_config(pipe:0000025B4378DB00, config:0000025B4378E700)
Rethrow as Exception: No device connected
Intel.RealSense.ErrorMarshaler.MarshalNativeToManaged (System.IntPtr pNativeData) (at <39ff202c67d64a63acbba4ae542e7167>:0)
Intel.RealSense.Pipeline.Start (Intel.RealSense.Config cfg) (at <39ff202c67d64a63acbba4ae542e7167>:0)
RsDevice.OnEnable () (at Assets/RealSenseSDK2.0/Scripts/RsDevice.cs:71)
---------------

Sambhav-Viztech, 2020-11-26:
RealSense Unity package works and so does the Viewer. Have not imported the rsense wrapper unity package into the Rsvfx project though, thanks again!