keijiro/Rsvfx #23: wiper vfx can not be edited in another project

fdiba, 2019-06-28:
Hello, when I try to edit wiper vfx inside another project the effect do not work anymore as soon as I open it. It's not the case for the other vfx (like particle, scanner or simple).

I can edit it in the original project but not in a new one. I get these errors: 

Remove 2 child(ren) that couldnt be deserialized from  of type UnityEditor.VFX.VFXGraph
UnityEditor.VFX.VFXGraph:OnEnable()

Remove 2 child(ren) that couldnt be deserialized from  of type UnityEditor.VFX.VFXGraph
UnityEngine.GUIUtility:ProcessEvent(Int32, IntPtr)

Any idea to solve the problem ?

keijiro, 2019-06-28:
I think that's because it uses some custom nodes from the [VfxExtra](https://github.com/keijiro/VfxExtra) project. I don't recommend using it in other projects. It's quite dirty hack so that it will be broken in future versions.

fdiba, 2019-06-28:
Ok, thank you for the tip. I hope the effect will be reproductible at a certain point without the VfxExtra project.

keijiro, 2019-06-28:
I'll do when VFX Graph gets a proper way to implement a custom node without a hacky way.

closed 2019-06-28 by keijiro