keijiro/Rsvfx #24: mapping the waveform to a particle system

fdiba, 2019-06-29:
Hello, I tried to map the waveform to a particle system in the vfx graph. The particle system is positioned in a AABox (size 10 0.1 10). He is responding to the input sound in the Y axis by going up and down but each particle is responding for the moment in the same way (by getting positioned at the same height).

I tried to use the x position of each particle to sample the waveform texture. The vfx effect is positionned at the origin of the world (0,0,0). I have divided the x position by 5 (because the aabox is 10 in the x axis), using remap to get a value between 0 and 1( remap(value,-1,1,0,1) ), and multiply this value by 1024 to get a diferent value for eac particle from the sample texture. What I am missing ?

You can find the vfx here: https://www.dropbox.com/s/72n05py86ldbxyr/Wavos.vfx?dl=0

fdiba, 2019-06-29:
Found it. It was an error to multiply everything by 1024. It would be nice to have a third example mapping the low, high and mid value on "waves of points".

keijiro, 2019-06-30:
Is this issue is related to this repository (Rsvfx)?

fdiba, 2019-06-30:
No it was related to the Rsvfx/Lasp repository sorry.

keijiro, 2019-07-01:
I'm closing this issue because it's not relevant to this repository. Please feel free to repost it to the correct repository, I'm not sure if it's actually an issue about the project though.

closed 2019-07-01 by keijiro