keijiro/Rsvfx #25: How to solve this problem？

Cicada3306, 2019-07-08:
Rethrow as Exception:Couldn't resolve request.

keijiro, 2019-07-08:
It seems to fail to connect to the RealSense device.

- Which RealSense device model are you using?
- Which platform (Windows/Mac/Linux) are you using?
- Which version of Unity are you using?
- Which scene in this project are you trying?

Cicada3306, 2019-07-08:
RealSense device：D415
using Windows
using Unity2019.1.8f
trying Simple scene

keijiro, 2019-07-08:
Thanks for the clarification. Could you check if the official Unity RealSense examples work on it? https://github.com/IntelRealSense/librealsense/releases/download/v2.24.0/Intel.RealSense.unitypackage

Cicada3306, 2019-07-08:
it also not work, i try Record and Playback still have this problem.

keijiro, 2019-07-08:
Sure. Then it's supposed that there is a problem with RealSense SDK or your environment/hardware. Please refer to the SDK documentation to solve the problem.

I'm closing this issue now. Please feel free to reopen it if it sill reproduces after solving the problem on the SDK side.

closed 2019-07-08 by keijiro

Cicada3306, 2019-07-08:
ok,thanks