keijiro/Rsvfx #27: Work with Spout

Zudva, 2019-08-14:
Hi! I want use you Asset with KlakSpout. Im not can use it in this project?

keijiro, 2019-08-15:
You can.

keijiro, 2019-08-15:
I'm closing this issue now, but please feel free to reopen for further questions.

closed 2019-08-15 by keijiro

Zudva, 2019-08-15:
Keijiro, i add Spout folder with  to Assets in your RsVFX project and add KlakSpout. Spout script see my Touchdesigner spout out, but Touchdesigner spout in node not see Unity Sender script for Camera in scene.
 Sorry, but im not understand how to correctly launch. In new project all work.
Thank you!

keijiro, 2019-08-15:
Please note that the camera capture mode in the Spout Sender component doesn't support SRP (LWRP/HDRP). That's why your project doesn't show anything on TouchDesigner side. I'd recommend using the Render Texture mode insted. Please check README of KlakSpout for further details.