keijiro/Rsvfx #29: Maximum distance from the camera

xubi1993, 2019-09-01:
Hello, I was playing around with this asset with intel realsense 435 camera and noticed that it only works when you are maximum 2 feet away from the camera once you go further than that it seems to stop detecting you. Is there a way to increase the detection range to something like 7 or 8 feet. Looking forward to your response and thank you for the great asset.

keijiro, 2019-09-01:
Tweak the Depth Threshold property in the PointCloudBaker component.

keijiro, 2019-09-01:
I'm closing this issue. Please feel free to reopen if the problem still reproduces.

xubi1993, 2019-09-04:
Hello again, I am having trouble getting the position data into the vfx graph. The problem is when I am near the camera it detects me and give me my silhouette in particle form but I want it to do the same when I am 3 meters away from the camera when I am at that distance it shows nothing Depth threshold has no effect on this. I am stuck can you please point me in the right direction to get it to work the same as it does when you are near the camera. Thank you.

keijiro, 2019-09-04:
Could you check if the depth threshold actually works on your environment? Stand at the point 1 meters away from camera, then decrease the depth threshold and check if the points disappear.

xubi1993, 2019-09-04:
Yes it is working as I turn its value down you can see yourself disappear from the scene but no matter how high I set it as soon as I start to move away from the camera after 1 feet I start to disappear and at around 2 feet I am completely invisible to the vfx graph.

keijiro, 2019-09-04:
I tested with D415. It seems that it can detects points at 2m or 3m away from the camera, the quality (accuracy) is not good though.

Could you test it with a simple effect, like Simple.unity in this repository? If it fails, then could you test it using realsense-viewer contained in the RealSense SDK?

xubi1993, 2019-09-05:
Hello, I have tested with simple effect, with plexus effect and with the real sense viewer as well and this is what I have got: [view](https://imgur.com/a/bwIRAAZ)

I was sitting at around 2.5m in all these shoots. As you can see I can see myself in the simple view but in effects view I see nothing maybe because there is to much noise in the background?

keijiro, 2019-09-05:
As shown in the screen shot of the Simple scene, the device detects objects (you, wall, desk, etc.) without a problem. It's very noisy, but it still works.

I think the effects (like Plexus) is not completely disappearing. It just gets very dark and sparse.

There are some problems to use these effects with this situation:

- The wall and the object is too close. It's hard to separate the object from the wall.
- The point cloud is too sparse to emit effects with enough density.
- Some effects use the color of the pixels. If you wear a black shirt, the effects will be invisible.

I think these problems can be resolved by optimizing the effects to the situation. Unfortunately, it's not in the scope of this project: This project is to provide a way to use RealSense with VFX Graph. These effects are just examples showing how to use the components.

I can give some advice, but please understand that I'm not in charge of resolving these problems.

xubi1993, 2019-09-05:
Yes I understand and thank you so much for your help and some good pointers I will look into them and how to tackle them.

closed 2019-09-05 by keijiro