keijiro/Rsvfx #31: Rsvfx doesn't use full frame data

codeKenner, 2019-10-29:
Hey Keijiro,
first of all, very nice work with the Rsvfx.
I noticed, that your implementation only uses part of the information captured by the Realsense camera. The frame seams to be cropped on all 4 edges when compared to the visual in the Intel Realsense Viewer. 
I did try 2 resolutions, 640x480px as well as 1280x720px. Setup works fine except for the crop. 
Any hints on that?

keijiro, 2019-10-30:
Which RealSense device are you using? D435 has a bit strange FoV setting, so it will be cropped when using color and depth at the same time. Please understand that it's "by-design". If you're using D415, this problem shouldn't be happened.

codeKenner, 2019-10-30:
Keijiro, thank you for the quick response. It‘s the D435 I‘m using. So I understand your point about it being „by design“. Can I use the depth info only with full info as a input into the vfx?

keijiro, 2019-10-30:
How about changing this line to “bool mask = true;” ? I can’t remember the details, but I think this mask thing does cropping.

keijiro, 2019-10-30:
Oops. Forgot to paste the URL.

https://github.com/keijiro/Rsvfx/blob/ab16a0dcb3daa41caf9eee5a16efe6e5c3878fb1/Assets/Rsvfx/Runtime/DepthConverter.compute#L46

keijiro, 2019-11-13:
I'm closing this issue now as the problem has been solved. Please feel free to reopen for further problems.

closed 2019-11-13 by keijiro