keijiro/Rsvfx #35: External Data Question

BloodRaine, 2020-04-15:
I had a question pertaining to external data usage, As long as the format of the data was the same (the point source and color source) would it be possible to stream external data from ROS using something like [ros-sharp](https://github.com/siemens/ros-sharp)? Or it the RealSense too integral to be able to use other data. For reference I'm trying to stream point cloud data from ROS into Unity and display the cloud in AR. I also know this is fairly similar to [this issue](https://github.com/keijiro/Rsvfx/issues/3) so I'm sorry if you are repeating any info here.

keijiro, 2020-04-15:
Rsvfx is too complicated as a reference for what you're trying to do. Most of the code is just for managing the RealSense device that is not very meaningful for your project.

It might sound strange, but I think Smrvfx is a better example for your project.

https://github.com/keijiro/Smrvfx

It consists of a few scripts and a simple compute shader. It just retrieves data and bakes it into textures.

https://github.com/keijiro/Smrvfx/tree/master/Assets/Smrvfx

keijiro, 2020-04-15:
I'm closing this issue now. Please feel free to reopen for further problems.

closed 2020-04-15 by keijiro

BloodRaine, 2020-04-15:
Thanks for the reply and the advice!