keijiro/Rsvfx #36: ARKit

smxthereisonlyone, 2020-05-27:
Would it be possible to adapt this effect to work on iOS using ARKit?

keijiro, 2020-05-28:
It needs lots of rewriting. It's almost the same as starting over -- not adapting.

closed 2020-05-28 by keijiro

smxthereisonlyone, 2020-06-02:
Thanks for your answer :)

But in theory it's possible and -  by your estimation - worthwhile? 

Would the Depth Map output of the iphones (maybe it varies across the generations?) be sufficient to achieve a stylized effect?

keijiro, 2020-06-02:
- I tried depth sampling with iPhone X a few years ago. The result was not good. It was far worse than RealSense, so I stopped working on it further.
- I expected that the LiDAR scanner of iPad Pro could capture high-quality raw depth images. As far as I know, Apple hasn't offered an API for it (they only provide a static mesh reconstruction API).

So my conclusion is that iOS doesn't have the capability for it.