keijiro/Rsvfx #38: No device connected when using D455

Sambhav-Viztech, 2020-11-25:
-using serial no. of device
-updatedpackages on package manager
-using unity 2019..4.14f1
-device firmware 5.12.9

keijiro, 2020-11-26:
The last update on this repository was made with Unity 2019.2.5f1. So, I'd recommend trying Unity 2019.2.x.

keijiro, 2020-11-26:
I'm closing this issue now. Please feel free to reopen it if it still reproduces.

closed 2020-11-26 by keijiro

Sambhav-Viztech, 2020-11-26:
Thank you for your swift response but still no luck on my end. Here is my error log:-
ExternalException: rs2_pipeline_start_with_config(pipe:0000025B4378DB00, config:0000025B4378E700)
Rethrow as Exception: No device connected
Intel.RealSense.ErrorMarshaler.MarshalNativeToManaged (System.IntPtr pNativeData) (at <39ff202c67d64a63acbba4ae542e7167>:0)
Intel.RealSense.Pipeline.Start (Intel.RealSense.Config cfg) (at <39ff202c67d64a63acbba4ae542e7167>:0)
RsDevice.OnEnable () (at Assets/RealSenseSDK2.0/Scripts/RsDevice.cs:71)

Sambhav-Viztech, 2020-11-26:
RealSense Unity package works and so does the Viewer. Have not imported the rsense wrapper unity package into the Rsvfx project though, thanks again!

Sambhav-Viztech, 2020-11-26:
Omg! Importing the latest rsense wrapper package into Rsvfx worked me Keijiro san! Thanks a ton for this awesome vfx project!

PierreDidier, 2021-10-05:
Hello Sambhav, I also have a realsense D455 and i'm currently trying to use Kejiiro's realsense vfx project. I opened it with the recommended version of unity (2019.2.5f1) and the good serial number but it still not working. Which version of rsens wrapper did you used and how have done to use properly Kejiiro's scene ? Thank you in advance