keijiro/Rsvfx #40: Update to 2020?

bryanrtboy, 2021-01-08:
Any chance of updating this project to 2020 or 2021?  This is the only project with SRP that I can run in Unity. The intel sample project only runs on much older versions of Unity.

keijiro, 2021-01-08:
I think you can run it on 2020/2021. Did you find any issue?

bryanrtboy, 2021-01-08:
I get an error ARBackgroundRenderer does not exist in the RsARBackgroundRenderer.cs script.  To fix this, I have to comment out the references to it. Importing ARFoundation does not work.

You're right though, most scenes with that commented out, except for the Capture scene that uses the SDK directly on the CombinedDriver.cs script. The error is the camera isn't recognized "ExternalException: rs2_pipeline_start_with_config"

keijiro, 2021-01-08:
Uh-oh. Okay, there is a compatibility issue on the SDK side. I should look into it.

masternader, 2021-01-08:
Same error (although I did not know other scenes are still functional with RsARBackgroundRenderer refs commented out) with capture scene.

Drumaddict182, 2021-03-12:
Hi keijiro, first thank you for your crazy work. I'm really interested of using Rsvfx but it doesn't work with 2020 unity. Is there a version of unity which can still run Rsvfx ? 

Arigato gozaimasu.

keijiro, 2021-03-13:
@Drumaddict182 I'm pretty sure that it runs on Unity 2019.2 or 2019.3. You can download them from https://unity3d.com/get-unity/download/archive

Drumaddict182, 2021-03-14:
Thanks you. I can open the project without error with Unity 2019.2.5. But when I load a scene and press play, the screen is all black, no video is displayed. Do you know what can be the reason ? I have a D415 Camera. 

Arigato gozaimasu.

Drumaddict182, 2021-03-15:
> 
> 
> Thanks you. I can open the project without error with Unity 2019.2.5. But when I load a scene and press play, the screen is all black, no video is displayed. Do you know what can be the reason ? I have a D415 Camera.
> 
> Arigato gozaimasu.

Ok so now it works and i didn't' change anything. Niiice !

getarobo, 2021-03-17:
Had similar issue on Unity 2020, I had set Occlusion Removal of RsProcessingProfile of Point Cloud to on or off.

Drumaddict182, 2021-05-05:
Hi everyone. Did someone find a way to run this project on Unity 2020 ? 

Thank you.

Drumaddict182, 2021-05-22:
@bryanrtboy  I'm trying to run this project on Unity 2020 but i have the same ARBackgroundRenderer issue. You say that you can fix this problem by comment out the references to it. But i don't understand which lines in the ARBackgroundRenderer.cs script i have to comment out... I'm not a developer. Any help would be very appreciated. Thanks a lot.

harryw123545, 2022-10-05:
Has anyone managed to find a fix for this yet?

Drumaddict182, 2022-10-06:
@harryw123545 I have managed to make it work with Unity 2021. It's the ARBackgroundRenderer.cs scipt that is the problem. I can share my project with you if you want. But you can not use the latest Realsense SDK with it, there is an other error. But it works fine with the Realsense SDK 2.3.

harryw123545, 2022-10-06:
@Drumaddict182 If you could share that would be very appreciated thank you!

harryw123545, 2022-10-06:
@Drumaddict182 Managed to get it working, thanks so much!