keijiro/Rsvfx #46: Device not detected while using D455 Model

yaqqolerz, 2024-04-26:
Hello, i was trying this project, my realsense model is D455, and when i pressed play it shows error device not detected. also i can only open it in 2019.2.5f1 editor (it would'nt change to 2020 or newer version), i tried to import the newest realsense sdk and it still doesn't do nothing, it just show another error when i play it. is there anything that i could do to fix this? thanks a lot!

keijiro, 2024-04-27:
Does it work with the official SDK samples?

yaqqolerz, 2024-04-27:
it does works with the sdk provided by intel

keijiro, 2024-04-27:
Does it mean `Intel.RealSense.unitypackage`?

yaqqolerz, 2024-04-27:
yes, i tried that package and there are samples in there, the device does works

keijiro, 2024-04-27:
Thanks for the clarification. Then the issue must be on my plugin side.

Unfortunately, I don't have any idea about the issue. I haven't touched this plugin for five years, so I can't remember anything about it.

yaqqolerz, 2024-04-27:
well that's unfortunate:( anyway can you give me suggestion on how to use the skeleton tracking provided by realsense in unity? because i tried to search multiple examples on the internet and so far none works or they're just showcasing the project. by far your project is the closest to what i need right now, thanks a lot!